MelbourneRoboCats5648/Reefscape_2025 #2: Create command based robot template project

jaebe2, 2025-01-05:
Use the 2025 WPILib to create a command based robot template project.
https://docs.wpilib.org/en/stable/docs/software/vscode-overview/creating-robot-program.html#creating-a-robot-program

Confirm that the template project builds successfully using the WPILib VSCode extension (no need to deploy the code).

Commit this template code (without any modifications) as the first commit on the Reefscape_2025 repository.

jaebe2, 2025-01-06:
Hi @whereisalyssa, since you have completed this task, you can add a comment saying that you have finished it and close this issue. Thanks, Josh

whereisalyssa, 2025-01-07:
the command based robot template has been created and commited

closed 2025-01-07 by whereisalyssa