MelbourneRoboCats5648/Reefscape_2025 #4: Determine what the Field of View of the LimeLight camera is

jaebe2, 2025-01-06:
We need to know the angular [Field of View](https://en.wikipedia.org/wiki/Field_of_view) of the LimeLight camera so that we can optimise the mounting position and mounting angle on the Robot.

The FOV will comprise a horizontal angle and vertical angle. The vertical FOV angle is more important for us to know.

This information could be obtained:
- directly from limelight documentation if it is available
- by calculation using sensor resolution and lens specifications
- or it might need to be directly measured by taking a picture using the limelight

Mikaela1685, 2025-01-07:
Found this link
https://docs.limelightvision.io/docs/docs-limelight/hardware/hardware-comparison
Pretty sure the FOV is H:62.5° V:48.9°

So horizontally 62.5 and vertically 48.9

jaebe2, 2025-01-11:
Hi @Mikaela1685 - you can close this issue since you have found the FOV using the docs. I've created a follow on issue to work out the optimal mounting angle position here: https://github.com/MelbourneRoboCats5648/Reefscape_2025/issues/16

closed 2025-02-22 by jaebe2