MelbourneRoboCats5648/Reefscape_2025 #11: Develop shootAndIntake cmd

soda-sann, 2025-01-11:
shooter and intake joint command

jaebe2, 2025-01-11:
@soda-sann - you might find this useful
https://docs.wpilib.org/en/stable/docs/software/commandbased/command-compositions.html

jaebe2, 2025-01-16:
Hi @soda-sann - managed to find the problems causing the build failure on your branch.

- Had to change the name of the IntakeAndShootCommand to PerformIntakeAndShootCommand since this function had the same name as the IntakeAndShootCommand class.
- There were also some virtual functions that needed to have an implementation (The Periodic and SimulationPeriodic functions).
- Also just a few typos in the way the IntakeAndShootCommand was written.

I've attached a git patch file with the changes from your branch to the version I got working. You can manually go through them by opening it in notepad++ and implementing the changes.

[bugfixPatchOnCommit_c0d2234_branch11.patch](https://github.com/user-attachments/files/18431759/bugfixPatchOnCommit_c0d2234_branch11.patch)

Alternatively if you are feeling adventurous, you can open the git bash command line on your branch and apply the patch using these instructions: https://gist.github.com/nepsilon/22bc62a23f785716705c

jaebe2, 2025-01-19:
Hi @soda-sann, Alyssa has made a few more changes on so it will be safe for you to merge in her branch again (branch 22) into yours. Once you have done this you can try building and deploying your changes on last seasons robot. If the intake and shooter are working as you expect then you can go ahead and raise a pull request.

closed 2025-01-25 by jaebe2