MelbourneRoboCats5648/Reefscape_2025 #17: skeleton of swerve drive code using command

Mikaela1685, 2025-01-11:


itsmevjnk, 2025-01-11:
You may want to have a look at this: https://github.com/CrossTheRoadElec/Phoenix6-Examples/tree/main/cpp/SwerveWithPathPlanner
Essentially we can use the Phoenix tuner software to set up our drivetrain, then put the generated config files in and use CTR's swerve drivetrain controller from the library to run (and simulate) the drivetrain. I have tested it in simulation, and it seems to also work with AdvantageScope and Path Planner.

jaebe2, 2025-01-11:
@Mikaela1685 - you might have already resolved this, but just in case you were still trying to get the pheonix libraries working, the 2025 WPILib has a new feature to update the 3rd party libraries:
https://docs.wpilib.org/en/stable/docs/software/vscode-overview/3rd-party-libraries.html

jaebe2, 2025-01-16:
Swerve drive base dimensions are now: 700mm x 700mm

itsmevjnk, 2025-01-16:
Hey @Mikaela1685 @jaebe2, I think I might have found the issue with the swerve drive code - it seems that Phoenix 6 has some breaking changes regarding `SetVoltage()` polarity (or it could be due to us upgrading the motors' firmware/running Phoenix Tuner X), causing the steer motors to turn in the wrong direction. This has been fixed in the new `wpilib-2025` branch of the Crescendo robot code: https://github.com/MelbourneRoboCats5648/Crescendo_2024/blob/wpilib-2025/src/main/cpp/DriveModule.cpp#L97

itsmevjnk, 2025-01-17:
@Mikaela1685 I just had a look at your code - the migration was correct, but there's a few things I want to add for DriveModule.cpp:
- On lines 43-44, `SupplyCurrentThreshold` and `SupplyTimeThreshold` still exist in the 2025 Phoenix 6 library - they're just renamed. `SupplyCurrentThreshold` is now `SupplyCurrentLowerLimit`, and `SupplyTimeThreshold` is now `SupplyCurrentLowerTime`.
- According to my testing last night on the Crescendo code, the shaky drive modules issue can be fixed by inverting the voltage setting for the direction motors. This is in line 77.

itsmevjnk, 2025-01-17:
We're still having problems with the robot code: it seems to never finish initialising. This is broken since commit `3a5f0215a7308bfb0e48ebc0f7e3c909cc6d22f4`, and it might be caused by the gyro object in DriveSubsystem, as removing it seems to fix the issue (even having the object without calibrating, reset or reading would cause it). Interestingly, this does not happen in the Crescendo code, which follows more or less the same initialisation procedure.

There are two options I can think of: either trying to fix this, or ditching the onboard gyroscope altogether and migrating to the Pigeon 2.0.

itsmevjnk, 2025-01-18:
Might be useful for creating controlled motion
https://github.wpilib.org/allwpilib/docs/release/cpp/classfrc2_1_1_swerve_controller_command.html