MelbourneRoboCats5648/Reefscape_2025 #27: limelight stoof

Mikaela1685, 2025-01-14:
limelight research

Mikaela1685, 2025-01-14:
examples on the limelight page but only java: https://github.com/LimelightVision/limelight-examples

jaebe2, 2025-01-14:
@Mikaela1685 
C++ limelight code example:
https://github.com/[LimelightVision/limelightlib-wpicpp](https://github.com/LimelightVision/limelightlib-wpicpp)

Mikaela1685, 2025-01-14:
there's this I found on the wayback machine that a bunch of cheif delphi people use but idk if it uses most recent libraries or not...
https://web.archive.org/web/20230227080728/https://docs.limelightvision.io/en/latest/cs_estimating_distance.html

Mikaela1685, 2025-01-14:
ok so i found these tutorials on the limelight page on april tags.
For like the reef we should be able to track the individual like pipes as distances from the april tag (assuming that they can be seen on the camera... idk if thats gonna be possible tho due to like positioning.)

jaebe2, 2025-01-16:
Might need to print this calibration board:
https://downloads.limelightvision.io/models/calib.io_charuco_200x150_8x11_15_12_DICT_5X5.pdf