MelbourneRoboCats5648/Reefscape_2025 #32: Add a timeout to the left climb subsystem

jaebe2, 2025-01-16:
Currently this subsystem is implemented by performing the climb action while controller button is pressed. Instead, we want to change the behaviour so that as soon as the button is pressed (perhaps use OnTrue instead of WhileTrue), the climb action occurs for exactly 1.5 seconds. 

Check how timeouts are implemented using this example:
https://docs.wpilib.org/en/stable/docs/software/commandbased/command-compositions.html

jaebe2, 2025-01-16:
Hi @musarrat-mahbub - i've added this for you to look at so that you can get a better understanding of composing more complex commands.

musarrat-mahbub, 2025-01-16:
I have added a timeout for the left climb and it built successfully. I noticed there wasn't an issue for the right climb subsystem - do you want me to also add it there?
```  
  m_joystick.Button(leftUpButton).OnTrue(m_leftClimbSubsystem.LeftClimbUpCommand().WithTimeout(1.5_s));
  m_joystick.Button(leftDownButton).OnTrue(m_leftClimbSubsystem.LeftClimbDownCommand().WithTimeout(1.5_s));
  m_joystick.Button(rightUpButton).WhileTrue(m_rightClimbSubsystem.RightClimbUpCommand());
  m_joystick.Button(rightDownButton).WhileTrue(m_rightClimbSubsystem.RightClimbDownCommand());
```

jaebe2, 2025-01-16:
Nice, I think before you do that we should test this on the robot if we can

musarrat-mahbub, 2025-01-16:
I fixed the code so that it actually works, and added it to right climb after testing. Please review my code.

jaebe2, 2025-01-16:
Hi @musarrat-mahbub - i think you might have forgotten to push your code. Once you have pushed it you can let me know and I can look at your branch. Alternatively since you have tested it on the robot and if you are happy with your changes, you can submit a pull request and i can review and add comments on your code. Thanks

musarrat-mahbub, 2025-01-17:
Sorry, I didn't see this message until now. I have pushed it now.

musarrat-mahbub, 2025-01-17:
I have fixed the typo and pushed my code

closed 2025-01-17 by jaebe2