MelbourneRoboCats5648/Reefscape_2025 #36: Add pose estimation to robot

jaebe2, 2025-01-17:
https://docs.wpilib.org/en/stable/docs/software/advanced-controls/state-space/state-space-pose-estimators.html

jaebe2, 2025-01-17:
Example code from [limelight docs](https://docs.limelightvision.io/docs/docs-limelight/pipeline-apriltag/apriltag-robot-localization) to do the pose estimation update:

` LimelightHelpers.PoseEstimate mt1 = LimelightHelpers.getBotPoseEstimate_wpiBlue("limelight");
      
      if(mt1.tagCount == 1 && mt1.rawFiducials.length == 1)
      {
        if(mt1.rawFiducials[0].ambiguity > .7)
        {
          doRejectUpdate = true;
        }
        if(mt1.rawFiducials[0].distToCamera > 3)
        {
          doRejectUpdate = true;
        }
      }
      if(mt1.tagCount == 0)
      {
        doRejectUpdate = true;
      }

      if(!doRejectUpdate)
      {
        m_poseEstimator.setVisionMeasurementStdDevs(VecBuilder.fill(.5,.5,9999999));
        m_poseEstimator.addVisionMeasurement(
            mt1.pose,
            mt1.timestampSeconds);
      }`

jaebe2, 2025-01-17:
Example pose estimation code using Mega Tag 2 (which is limelight's pose estimation algorithm):
https://docs.limelightvision.io/docs/docs-limelight/pipeline-apriltag/apriltag-robot-localization-megatag2

Mikaela1685, 2025-01-19:
https://docs.limelightvision.io/docs/docs-limelight/tutorials/tutorial-swerve-pose-estimation

jaebe2, 2025-02-22:
This has been implemented on branch #58 starting path planner
This issue can be closed once branch #70 has been merged into develop