MelbourneRoboCats5648/Reefscape_2025 #37: Add PID control to the 2024 climb subsystem

jaebe2, 2025-01-17:
The right climb subsystem will need a way of controlling its position. The 2024 robot doesn't have any encoders on its lift system, so this won't be immediately testable.

The aim will be to push a controller button that will move the climb to a set position. A WPI PID controller will be for this: https://docs.wpilib.org/en/stable/docs/software/advanced-controls/controllers/pidcontroller.html#

This example code could be useful:
https://github.com/wpilibsuite/allwpilib/tree/main/wpilibcExamples/src/main/cpp/examples/ElevatorTrapezoidProfile

This will follow on from issue #23 which can be used as the branch to implement this change.

musarrat-mahbub, 2025-01-17:
PID Control in Command based:
https://docs.wpilib.org/en/stable/docs/software/commandbased/pid-subsystems-commands.html#pid-control-in-command-based

musarrat-mahbub, 2025-01-17:
Combining Motion Profiling and PID in Command-Based
https://docs.wpilib.org/en/stable/docs/software/commandbased/profilepid-subsystems-commands.html

Kamryn-123, 2025-01-18:
working on PID Controller in branch 23. Both for elevator subsystem. Controller is nearly finished, builds successfully rn.

jaebe2, 2025-01-19:
@musarrat-mahbub, @Kamryn-123, @whereisalyssa - This might be a useful video when we get to tuning the PID controller on the prototype elevator:
https://docs.wpilib.org/en/stable/docs/software/advanced-controls/introduction/pid-video.html

Kamryn-123, 2025-01-20:
replicated all of the changes from branch #38. Code builds!

Kamryn-123, 2025-02-07:
hello, this issue doesn't have an allocated branch as all the PID changes were made on #38. PID control has now also been added to the elevator. Are we able to close this issue?

jaebe2, 2025-02-08:
Hi @Kamryn-123 - yes you can close this issue

closed 2025-02-09 by Kamryn-123