MelbourneRoboCats5648/Reefscape_2025 #38: Add command based PID control to the left climb subsystem

jaebe2, 2025-01-17:
Add PID controller to control the height of the left climb similar to issue #37

Use a xbox controller button to trigger the climb to reach a set point. The spark max motor controllers will need to be added to the subsystem similar to branch #23

musarrat-mahbub, 2025-01-17:
some useful stuff:
PID Control in WPILib:
https://docs.wpilib.org/en/stable/docs/software/advanced-controls/controllers/pidcontroller.html#
PID Control in Command based:
https://docs.wpilib.org/en/stable/docs/software/commandbased/pid-subsystems-commands.html#pid-control-in-command-based

jaebe2, 2025-01-18:
Hi @musarrat-mahbub, the electrical team have created a prototype to test PID control using the rev spark max controller and the new rev motors. We haven't tested it out yet but might be fairly close.
@Kamryn-123 has been making some progress on the implementation of this today along with @itsmevjnk. If you are coming in tomorrow you can have a chat with Kam so that you can get up to speed before you continue coding on your branch. Its okay if your implementation will end up identical to hers as long as you try to understand what the code is doing while you write it.

jaebe2, 2025-01-19:
Hi @musarrat-mahbub - vinh and I have added a few commits on your branch and have it building. Have a look and see if you understand what the changes we made were.
@Kamryn-123 - take a look at these commits (on branch 38-temp-bug-fix) and see if you can replicate them on your temporary bug fix branch to resolve your build error. Don't merge musarrat's branch branch into your branch however.

After this once Alyssa has merged her elevator change into main, you can pull in main into your branch and migrate your controller code into the elevator subsystem so that we can test it. I can go through this part with you after I get there.

jaebe2, 2025-01-23:
I've added a patch file of the above changes to make it easier to copy past if you wanted:

[pidControllerPatch_onBranch38_commit_cecdf12.patch](https://github.com/user-attachments/files/18516186/pidControllerPatch_onBranch38_commit_cecdf12.patch)

itsmevjnk, 2025-01-23:
Hi @jaebe2 - while it is true that the goal is not set properly, I believe `LeftClimbSubsystem::LeftClimbL1Command` should be kept to have a `units::turn_t` input, since we'll always be stopping at the goal instead of proceeding past it. Also, `LeftClimbL1Command` is a bit of a misnomer, since it can be set to any goal and not just the L1 goal - I suggest changing it to `LeftClimbCommand` instead.

itsmevjnk, 2025-01-23:
Here are my suggested changes:

[image]

And here's the corresponding patch file contents:
```
diff --git a/src/main/cpp/RobotContainer.cpp b/src/main/cpp/RobotContainer.cpp
index 3075432..736e2aa 100644
--- a/src/main/cpp/RobotContainer.cpp
+++ b/src/main/cpp/RobotContainer.cpp
@@ -41,7 +41,7 @@ void RobotContainer::ConfigureBindings() {
   m_joystick.Button(rightDownButton).OnTrue(std::move(m_rightClimbSubsystem.RightClimbDownCommand()).Repeatedly().WithTimeout(1.5_s));
 
 // PID climb subsystem command (temporarily disabling LeftClimbUpCommand to test it)
-  m_joystick.Button(leftUpButton).OnTrue(m_leftClimbSubsystem.LeftClimbL1Command(1.0_tr));
+  m_joystick.Button(leftUpButton).OnTrue(m_leftClimbSubsystem.LeftClimbCommand(m_climbGoalL1));
 
   // Schedule `ExampleMethodCommand` when the Xbox controller's B button is
   // pressed, cancelling on release.
diff --git a/src/main/cpp/subsystems/LeftClimbSubsystem.cpp b/src/main/cpp/subsystems/LeftClimbSubsystem.cpp
index ab8ee5b..60206ac 100644
--- a/src/main/cpp/subsystems/LeftClimbSubsystem.cpp
+++ b/src/main/cpp/subsystems/LeftClimbSubsystem.cpp
@@ -60,9 +60,11 @@ frc2::CommandPtr LeftClimbSubsystem::LeftClimbDownCommand() {
           .FinallyDo([this]{m_motorClimbLeft.Set(0);});
 }
 
-frc2::CommandPtr LeftClimbSubsystem::LeftClimbL1Command(units::turn_t goal) {
-  return Run([this] {
-          m_leftClimbSetpoint = m_TrapezoidalProfile.Calculate(LeftClimbConstants::kDt, m_leftClimbSetpoint, m_leftClimbGoal);
+frc2::CommandPtr LeftClimbSubsystem::LeftClimbCommand(units::turn_t goal) {
+
+  return Run([this, goal] {
+          frc::TrapezoidProfile<units::turn>::State goalState = { goal, 0.0_tps }; // stop at goal
+          m_leftClimbSetpoint = m_TrapezoidalProfile.Calculate(LeftClimbConstants::kDt, m_leftClimbSetpoint, goalState);
           m_closedLoopController.SetReference(m_leftClimbSetpoint.position.value(), SparkLowLevel::ControlType::kPosition);
         })
         .FinallyDo([this]{m_motorClimbLeft.Set(0);});
diff --git a/src/main/include/RobotContainer.h b/src/main/include/RobotContainer.h
index d92177a..03935cd 100644
--- a/src/main/include/RobotContainer.h
+++ b/src/main/include/RobotContainer.h
@@ -43,5 +43,7 @@ class RobotContainer {
   ShooterSubsystem m_shooterSubsystem;
   ElevatorSubsystem m_elevatorSubsystem;
 
+  units::turn_t m_climbGoalL1 = 1.0_tr; // TODO: we may want to move this to Constants.h or something
+
   void ConfigureBindings();
 };
diff --git a/src/main/include/subsystems/LeftClimbSubsystem.h b/src/main/include/subsystems/LeftClimbSubsystem.h
index 69885d8..288fd38 100644
--- a/src/main/include/subsystems/LeftClimbSubsystem.h
+++ b/src/main/include/subsystems/LeftClimbSubsystem.h
@@ -71,7 +71,7 @@ class LeftClimbSubsystem : public frc2::SubsystemBase {
    */
   frc2::CommandPtr LeftClimbUpCommand();
   frc2::CommandPtr LeftClimbDownCommand();
-  frc2::CommandPtr LeftClimbL1Command(units::turn_t goal);
+  frc2::CommandPtr LeftClimbCommand(units::turn_t goal);
 
   /**
    * Will be called periodically whenever the CommandScheduler runs.
```

jaebe2, 2025-01-23:
Hi @itsmevjnk, yes your changes are good. Its better to just set the goal as turn_t and hide the goal state from the calling function.
Correct about the naming as well when we eventually do merge the changes in. For now this was just to test that PID control works as we think it does. Might be good to test the changes on the prototype board if it is still available.

itsmevjnk, 2025-01-23:
Hi @jaebe2, yesterday we tested both your code and mine, and they all worked as expected - however, we had to disable the elevator subsystem for now, since it was also set to use the same motor controller and would cause conflicts at runtime. This will not be an issue once we migrate the left climb subsystem to the elevator subsystem, at which point the left climb subsystem will be gone.

closed 2025-02-02 by jaebe2