MelbourneRoboCats5648/Reefscape_2025 #43: Troubleshooting the Spark Max brushless DC motor controllers

jaebe2, 2025-01-20:
- Create a 2025 WPI timed robot example project.
- Add the Rev vendor libraries (issue #21) and include the relevant rev/SparkMax.h header files.
- Create a SparkMax motor controller and xbox controller as a member of robot.h and configure it as required (set correct CAN ID etc.).
- Deploy code on the RoboRio mounted on the prototype elevator setup
- See if the motor can be set to spin when a xbox controller input is pressed (use previous season's code as guidance).

This code shouldn't be commited, its just for testing.
Try a few different things in the code to see if the motor controller starts working.

whereisalyssa, 2025-01-21:
the motor controller works with the elevator subsystem (we didnt save the configured settings for the spark max - therefore wrong can id)

jaebe2, 2025-01-21:
@whereisalyssa - awesome, you can close this issue

closed 2025-01-21 by whereisalyssa