MelbourneRoboCats5648/Reefscape_2025 #55: Tune swerve drive parameters

jaebe2, 2025-02-01:
May need to optimise parameters to make swerve more responsive and reduce jitter:

- [ ] Turn PID controller
- [ ] Drive PID controller (check if the feedforward has been tuned correctly)
- [ ] Slew rate limiters

Mikaela1685, 2025-02-08:
IC PID Values
 Normal Drive = 
  const double TURN_P = 3; (CHECK IF WE CONFIG OUR TURN MOTOR)
  const double TURN_I = 0.0;
  const double TURN_D = 0;
  const double DRIVE_P = 0.37586; //0.000031489;
  const double DRIVE_I = 0.0;
  const double DRIVE_D = 0.0;
  const double DRIVE_F = 0;

PathPlanner =
          PIDConstants(2, 0.0, 0.0),    // Translation PID constants
          PIDConstants(0.5, 0.0, 0.0),  // Rotation PID constants
          MAX_VELOCITY,                 // Max module speed, in m/s
          432_mm,  // Drive base radius in meters. Distance from robot center to furthest module.
                   // NEEDS TO BE CHECKED AND MADE ACCURATE!!

Mikaela1685, 2025-02-08:
Looking at utilities/swervmodule.cpp I don't see anywhere where we config our turn motor. (someone else should probs do a more thorough check.)
in the case we don't, IC's turn motor config is:
(they call the function inside there config thing which is probs not necessary... or probs better... idk)

void SwerveModule::ConfigTurnMotor(){
    // Config Turning Motor
 // _canTurnMotor.RestoreFactoryDefaults();
  //_canTurnMotor.SetCANTimeout(500);
  _canTurnMotor.SetConversionFactor(1.0 / TURNING_GEAR_RATIO);
  _canTurnMotor.EnableClosedLoopWrapping(0_tr, 1_tr);
  _canTurnMotor.SetPIDFF(TURN_P, TURN_I, TURN_D);
  _canTurnMotor.SetInverted(true);
  _canTurnMotor.SetIdleMode(rev::CANSparkMax::IdleMode::kBrake);
  //_canTurnMotor.BurnFlash();
  //_canTurnMotor.SetCANTimeout(10);
}

Mikaela1685, 2025-02-08:
Also Idk if this will be helpful at all but...
https://www.reddit.com/r/FRC/comments/r6siuy/pid_tuning/
https://www.youtube.com/watch?v=jIKBWO7ps0w

itsmevjnk, 2025-02-08:
> IC PID Values Normal Drive = const double TURN_P = 3; (CHECK IF WE CONFIG OUR TURN MOTOR) const double TURN_I = 0.0; const double TURN_D = 0; const double DRIVE_P = 0.37586; //0.000031489; const [...]

We might actually test this tomorrow

itsmevjnk, 2025-02-15:
Apparently we're supposed to also tune the feedforward parameters for the speed motors, and to do so **before** tuning its PID. CTRE docs might come in handy: https://v6.docs.ctr-electronics.com/en/2024/docs/api-reference/device-specific/talonfx/closed-loop-requests.html

We should be able to do so after #65 is addressed.

itsmevjnk, 2025-02-15:
These parameters seem to be a good start:
* kP = 0.25, kI = 0.0, kD = 0.0
* kV = 2.65, kS = 0.2, kG = 0.0
They seem to work fine on the 2024 robot running on the foam mat, but the robot was going up to only 0.5m/s.
A few things to note here:
* Once we get the actual speed to match the commanded speed, smaller increases in kV seems to reduce the lag between commanded and actual speed. Larger increases, however, would cause the latter to overshoot.
* It looks like kI and kD should be left at zero - feedforward alone seems to be enough, and kP also helps with lag reduction (while also causing the actual speed to undershoot, so we might need to increase kV as we tune kP).
* kG is to be set to zero, since we're not working against gravity here (unlike arms and elevators).