MelbourneRoboCats5648/Reefscape_2025 #60: Add cosine scaling to swerve drive

jaebe2, 2025-02-02:
Example SwervePoseEstimator code has implementation of cosine correction during swerve turns. This might be helpful to reduce jitter during swerve manouvres.

```
  // Scale speed by cosine of angle error. This scales down movement
  // perpendicular to the desired direction of travel that can occur when
  // modules change directions. This results in smoother driving.
  referenceState.CosineScale(encoderRotation);
```

jaebe2, 2025-02-02:
The above code is from the SwervePoseEstimator example code from WPI lib. It would be helpful to create this project to see where this cosine scaling is happening and try to implement this in our swerve drive code.

musarrat-mahbub, 2025-02-08:
I have added cosine scaling, but I'm not sure how to add a button to toggle it on/off

itsmevjnk, 2025-02-08:
@jaebe2 We've tested the code with and without cosine scaling (by commenting the code out), and cosine scaling actually kind of helped with the initial swerving caused when the wheels change direction - not by much but there's an improvement.

For now I'll create a PR for this, but I want to run it through you first, in case there's something else you want to add.

closed 2025-02-08 by itsmevjnk