MelbourneRoboCats5648/Reefscape_2025 #76: Implement drivetrain point to point motion

jaebe2, 2025-02-16:
The following example for following a trajectory uses a ramesete controller which is only used for differential drives:
https://docs.wpilib.org/en/stable/docs/software/pathplanning/trajectory-tutorial/creating-following-trajectory.html

The following swerve controller command could be used instead of the ramesete controller for a swerve drive:
https://github.wpilib.org/allwpilib/docs/release/cpp/classfrc2_1_1_swerve_controller_command.html
https://docs.wpilib.org/en/stable/docs/software/advanced-controls/trajectories/holonomic.html

Mikaela1685, 2025-02-22:
code from:
https://github.com/wpilibsuite/allwpilib/blob/main/wpilibcExamples/src/main/cpp/examples/SwerveControllerCommand/include/RobotContainer.h

jaebe2, 2025-04-27:
@Mikaela1685 and @itsmevjnk - could possible resurrect this branch for auto stuff

jaebe2, 2025-06-28:
Implementation of holomonic trajectory controller in C++

https://github.com/TEAM1771/Black-Widdow-2022-Update/blob/master/src/main/cpp/Trajectory.cpp

jaebe2, 2025-07-05:
Example holonomic controller. Compiles

[image]