MelbourneRoboCats5648/Reefscape_2025 #115: Switch between field-centric and robot-centric drive controls

itsmevjnk, 2025-03-03:
We need to bind field/robot-centric toggle to a button, so that drive team can switch between these modes during matches.

Additionally, we might want to output the drive control mode to a DIO pin, so that we can switch the LED strip on and off.

itsmevjnk, 2025-03-03:
One more thing that I noticed is that GetHeading() is reporting the wrong value, causing field-centric to not work. To fix this, we'll need to invert the value - this has been tested on the robot today, but I haven't committed the change.

itsmevjnk, 2025-03-03:
@jaebe2 - shall we discuss this tomorrow?

jaebe2, 2025-03-03:
> [@jaebe2](https://github.com/jaebe2) - shall we discuss this tomorrow?

Yep, lets have a look at this tomorrow. The LED is a nice to have to help the drivers but not the highest priority. Getting the elevator tuned and auto placing as much as possible is probably most important

jaebe2, 2025-03-03:
I'm wondering if the heading issue is because I told Adeepa to mount the pigeon upside down, I thought this was the correct mounting based on the pigeon X axis pointing to the right, Y pointing straight which results in Z pointing up. I could be wrong.

itsmevjnk, 2025-03-03:
According to [WPILibC++ API](https://github.wpilib.org/allwpilib/docs/release/cpp/structfrc_1_1_chassis_speeds.html#abf60209b30cdef24c3a2a67f2c0cd62a), the heading to be supplied into the field-centric conversion function is counterclockwise positive. However, if I remember correctly, `GetAngle()` from the Pigeon is clockwise positive instead (there's a build time warning about that). This is probably what went wrong.

Since `GetAngle()` is deprecated anyway, the proper way to fix this is to change it to something like `GetYaw()` (once again, this is stated in the warning message).

jaebe2, 2025-04-27:
Implemented and merged into main via 4677e51

closed 2025-04-27 by jaebe2