MelbourneRoboCats5648/Reefscape_2025 #118: Switch to using ProfiledPIDController instead of Spark Max + TrapezoidProfile hybrid

itsmevjnk, 2025-03-04:
This may potentially resolve the issue where trapezoid profiling ends earlier than the mechanism's actual movement, causing weird behaviours.

itsmevjnk, 2025-03-04:
@jaebe2 - we might bring the changes over to the other branches and test if it works better at comp

jaebe2, 2025-03-04:
Had a quick look and code change doesn't seem too significant, however i'm a litte concerned that tuning new feedforward params would be too time consuming before comp though

itsmevjnk, 2025-03-04:
Our controller is predominantly feedforward, which we haven't swapped out - so we don't have to change feedforward params. If anything, we might only need to change PID, but that's simple.

jaebe2, 2025-04-27:
Implemented and merged into main in b16c9ee

closed 2025-04-27 by jaebe2