MelbourneRoboCats5648/Reefscape_2025 #133: photon vision pose estimation

itsmevjnk, 2025-05-18:
### Discussed in https://github.com/MelbourneRoboCats5648/Reefscape_2025/discussions/132

<div type='discussions-op-text'>

<sup>Originally posted by **Mikaela1685** May 18, 2025</sup>
https://docs.photonvision.org/en/latest/docs/programming/photonlib/robot-pose-estimator.html
https://github.com/PhotonVision/photonvision-docs/blob/master/source/docs/programming/photonlib/robot-pose-estimator.rst#id3
https://github.com/PhotonVision/photonvision/blob/main/photonlib-cpp-examples/poseest/src/main/cpp/subsystems/SwerveDrive.cpp
https://cppdocs.photonvision.org/release/classphoton_1_1_photon_pose_estimator.html

This looks interesting and good for testing and stuff:
https://www.chiefdelphi.com/t/photonvision-simulation-inconsistent-pose-estimation/457925

Apriltag field layout thing
https://github.wpilib.org/allwpilib/docs/release/cpp/classfrc_1_1_april_tag_field_layout.html
https://github.wpilib.org/allwpilib/docs/release/cpp/namespacefrc.html#a5f23885bcc9e850c6b99c8cd48b98fd9
</div>

jaebe2, 2025-05-18:
We will need to copy the method to get the standard deviations to add to the vision measurement using the code here:
https://github.com/PhotonVision/photonvision/blob/main/photonlib-cpp-examples/poseest/src/main/include/Vision.h

Mikaela1685, 2025-06-22:
Add a function to automatically align the robot to an apriltag on the reef.
This will require the robot to control its translation, and rotation with respect to the target.